intel/drivers.gpu.control-library #16: [Q] Get driver version

DevTechProfile, 2022-07-13:
How can get the driver version (like 30.0.101.1743) from the CTL? "ctl_version_info_t"  via "ctlCheckDriverVersion" is only an uint type.

IAmAnkur, 2022-07-14:
Please refer PrintAdapterProperties function in Sample_ControlAPP.cpp

you need to do ctlGetDeviceProperties(hDevices[Index], &StDeviceAdapterProperties) call and will get the version detail in "StDeviceAdapterProperties.driver_version"

DevTechProfile, 2022-07-20:
Reopened this because driver version via "StDeviceAdapterProperties.driver_version" is 8444249307940559. Difficult to get to  30.0.101.1743 from 8444249307940559.

ph0b, 2022-07-20:
could this be documented in igcl_api.h directly? 
8444249307940559 = 0x1E0000006506CF
0x001E=30
0x0000=0
0x0065=101
0x06CF=1743

this is actually implemented here: https://github.com/intel/drivers.gpu.control-library/blob/df9a2df0354e6a2b65900792a5e80df5cd531912/Samples/Generic_Sample/Sample_ControlAPP.cpp#L607

DevTechProfile, 2022-07-20:
Thanks, that wasn't far away...

closed 2022-07-20 by DevTechProfile