intel/drivers.gpu.control-library #28: ctl_mem_bandwidth_t usage unclear

tycho, 2022-10-17:
The comments in here:

https://github.com/intel/drivers.gpu.control-library/blob/de1d5b62bc973236c14b8de6e770cc59787124d5/includes/igcl_api.h#L5101-L5122

Suggest there should be readCounter/writeCounter members to this structure, but they aren't present. What's the intended way to use this structure?

mravichan, 2022-12-01:
IGCL changes are already merged to CI master , CI master : [gfx-driver-ci-master-12848](https://ubit-gfx.intel.com/build/15629428/overview) and fix is part of Prod IGCL 127.

ezhilintc, 2023-04-14:
@tycho -- The API has been updated in recent releases. Can you verify and agree to close this issue?

ashutosh-d-shukla, 2023-04-18:
This is already fixed as mentioned in the above comments, closing the issue.

closed 2023-04-18 by ashutosh-d-shukla