intel/drivers.gpu.control-library #33: Invalid telemetry data for Arc mobile

DevTechProfile, 2022-12-07:
Our telemetry implementation works fine for Arc desktop cards like A770. 

Code: https://github.com/CXWorld/CapFrameX/blob/release/1.7.1/source/CapFrameX.IGCL/IGCLManager.cpp

But for a mobile GPU like A770M the sensors calculated via (current - prev)/deltatime (TDP, loads) are invalid.

[image]

ezhilintc, 2023-01-09:
Please make sure, the adapter queried is accurate, as there might be multiple Intel adapters (integrated and discrete) in mobile platforms. Could you please specify, which API's are reporting incorrect Values?

DevTechProfile, 2023-01-09:
We have a filter for iGPUs so we are talking about sensors of the A770M. For example, the GPU TDP sensor is wrong. 

Your code:
`
ctl_result_t status = ctlPowerTelemetryGet(hDevices[index], &pPowerTelemetry);

if (status == ctl_result_t::CTL_RESULT_SUCCESS)
{
	prevtimestamp = curtimestamp;
	curtimestamp = pPowerTelemetry.timeStamp.value.datadouble;
	deltatimestamp = curtimestamp - prevtimestamp;

	if (pPowerTelemetry.gpuEnergyCounter.bSupported)
	{
		telemetryData->gpuEnergySupported = true;
		prevgpuEnergyCounter = curgpuEnergyCounter;
		curgpuEnergyCounter = pPowerTelemetry.gpuEnergyCounter.value.datadouble;

		telemetryData->gpuEnergyValue = (curgpuEnergyCounter - prevgpuEnergyCounter) / deltatimestamp;
	}
}
`
It works fine with the A770 desktop card.

DevTechProfile, 2023-02-07:
@pmmeyer 

1. All sensors calculated by (current - prev)/deltatime (TDP, loads) are invalid.
2. 1s refresh rate
3. Idle and under load

Thanks for looking at this.

pmmeyer, 2023-02-13:
@DevTechProfile  - can you please confirm which graphics driver you are using?

DevTechProfile, 2023-02-14:
Yes, on the left side you find the used driver version.

[image]

pmmeyer, 2023-03-16:
This appears to be an issue with CapFrameX's handling of the two display adapters.

Using the same `ctlPowerTelemetryGet` API, I wrote a sample application to log power and render activity for both the discrete and integrated display adapters. I ran my app and also used CapFrameX's capture tool to log the same metrics while running 3DMark TimeSpy.

Here is the output from my app:
[image]

CapFrameX's data are invalid though - negative numbers and very large numbers that are junk.

One thing I noticed is that for a given metric (e.g. power) the iGPU and dGPU numbers are almost identical.

Another thing I noticed was that if I disabled the integrated adapter in device manager, launched CapFrameX so that it only detected the discrete adapter, and then re-enabled the integrated adapter, I was able to run TimeSpy and get good data reported by CapFrameX for the discrete adapter. But whenever launching CapFrameX with both adapters enabled, the data get corrupted. 

I would start debugging by logging all of the telemetry returned by the API and manually computing power/activity to validate that the data by the API is valid and then dig deeper to see if that data is somehow being copied/accessed by the wrong adapter when processing.

DevTechProfile, 2023-03-21:
Many thanks @pmmeyer, I will check this soon.

DevTechProfile, 2023-03-24:
I think I could fix the issue.

closed 2023-03-24 by DevTechProfile