intel/drivers.gpu.control-library #34: ctlEnumerateDevices returns adapter count of zero for Intel UHD enable hardware

Twon, 2022-12-24:
Running the samples on a laptop with (Intel® UHD Graphics for 11th Gen Intel® Processors)[https://ark.intel.com/content/www/us/en/ark/products/213803/intel-core-i711800h-processor-24m-cache-up-to-4-60-ghz.html] ctlEnumerateDevices turns an adapter count of zero. Can you confirm the library should see this, and if not why not?  Many thanks.

rahulrgandhi99, 2023-01-09:
Control library is not supported on [Intel® UHD Graphics for 11th Gen Intel® Processors](https://ark.intel.com/content/www/us/en/ark/products/213803/intel-core-i711800h-processor-24m-cache-up-to-4-60-ghz.html) that is why it is reporting 0 adapters but we fixed this It will report unsupported platform instead of 0 adapters in latest drops.

paulober, 2023-01-09:
But does it then return for `graphics_adapter_properties`:
```
enumerator CTL_ADAPTER_PROPERTIES_FLAG_INTEGRATED:
[out] Is Integrated Graphics adapter
```

rahulrgandhi99, 2023-01-11:
for Arc dGPUs it won't set this flag **enumerator CTL_ADAPTER_PROPERTIES_FLAG_INTEGRATED:
[out] Is Integrated Graphics adapter** as it is not internal graphics adapter.

Twon, 2023-01-12:
> Control library is not supported on [Intel® UHD Graphics for 11th Gen Intel® [...]

Thanks, that's a more helpful output and I see that on the latest version of the drivers.   

However, as a user, it was not clear to me on encountering this error as to if this should work or not. Is it documented anywhere which systems this library is supported on?  Also is there a road map documenting which systems are going to be supported in the future?  Without this information hard to assess the value of supporting this library.

rahulrgandhi99, 2023-01-12:
Please find information about platform support at https://intel.github.io/drivers.gpu.control-library/prog.html#system-requirements

ashutosh-d-shukla, 2023-04-18:
Closing this as the query is already addressed in the comments.

closed 2023-04-18 by ashutosh-d-shukla