intel/drivers.gpu.control-library #39: result data not found when ctlGetSet3DFeature for app

paulober, 2023-01-09:
I get a `CTL_RESULT_ERROR_DATA_NOT_FOUND` when trying to read settings for a game. I set appname to the name of the exe like in the sample. For example "GTA.exe".  And i prviously added the game to Arc Control and changed the setting i now want to read for this app. 
Is there any way to get all the app specific settings which has been made?

rahulrgandhi99, 2023-01-11:
can you please paste your set and get code here?

paulober, 2023-01-13:
> can you please paste your set and get code here?

```cpp
if (h_api_handle_ == nullptr || *adapter_count_ < 1 || selected_device_ < 0)
{
    return false;
}

ctl_3d_feature_getset_t set_3d_property{ 0 };
set_3d_property.Size = sizeof(ctl_3d_feature_getset_t);

set_3d_property.FeatureType = CTL_3D_FEATURE_GAMING_FLIP_MODES;
set_3d_property.bSet = true;
set_3d_property.CustomValueSize = 0;
set_3d_property.pCustomValue = nullptr;

// application specify
if (application != nullptr)
{
    //app_name is of type: const char* app_name
    set_3d_property.ApplicationName = const_cast<char*>(app_name);
    set_3d_property.ApplicationNameLength = static_cast<int8_t>(strlen(app_name));
}

set_3d_property.ValueType = CTL_PROPERTY_VALUE_TYPE_ENUM;
set_3d_property.Value.EnumType.EnableType = static_cast<ctl_gaming_flip_mode_flag_t>(flip_mode);
set_3d_property.Version = 0;

if (h_devices_ != nullptr)
{
    WRITE_LINE("Trying to set CTL_3D_FEATURE_GAMING_FLIP_MODES to "
        + set_3d_property.Value.EnumType.EnableType);
    
    if (const ctl_result_t result = ctlGetSet3DFeature(h_devices_[selected_device_], &set_3d_property);
        result == CTL_RESULT_SUCCESS)
    {
        return true;
    }

    WRITE_LINE("Failed to set 3d feature of type CTL_3D_FEATURE_GAMING_FLIP_MODES.");
}

return false;
```

rahulrgandhi99, 2023-01-20:
> 

This is only set code. seems correct. Please paste your get code as well as you are facing ICTL_RESULT_ERROR_DATA_NOT_FOUND when you are trying to read settings for a game.

paulober, 2023-01-20:
Ok, here is the get code:
```c
if (h_api_handle_ == nullptr || *adapter_count_ < 1 || selected_device_ < 0)
{
    return CTL_GAMING_FLIP_MODE_FLAG_MAX;
}

ctl_3d_feature_getset_t get_3d_property{0};
get_3d_property.Size = sizeof(ctl_3d_feature_getset_t);

get_3d_property.FeatureType = CTL_3D_FEATURE_GAMING_FLIP_MODES;
get_3d_property.bSet = false;
get_3d_property.CustomValueSize = 0;
get_3d_property.pCustomValue = nullptr;
// application specify
if (application != nullptr)
{
    // app_name is of type: const char* app_name
    get_3d_property.ApplicationName = const_cast<char*>(app_name);
    get_3d_property.ApplicationNameLength = static_cast<int8_t>(strlen(app_name));
}
get_3d_property.ValueType = CTL_PROPERTY_VALUE_TYPE_ENUM;
get_3d_property.Version = 0;

if (h_devices_ != nullptr)
{
    if (const ctl_result_t result = ctlGetSet3DFeature(h_devices_[selected_device_], &get_3d_property);
        result == CTL_RESULT_SUCCESS)
    {
        const auto gfm = static_cast<ctl_gaming_flip_mode_flag_t>(get_3d_property.Value.EnumType.EnableType);
        return gfm;
    }
    else
    {
        WRITE_LINE("error reading global gaming flip mode! Result: "
            + decode_ret_code(result).c_str());
    }
}

return CTL_GAMING_FLIP_MODE_FLAG_MAX;
```

rahulrgandhi99, 2023-01-23:
> Ok, here is the get code:
>
> ```c
> if (h_api_handle_ == nullptr || *adapter_count_ < 1 || selected_device_ < 0)
> {
>     return CTL_GAMING_FLIP_MODE_FLAG_MAX;
> }
>
> ctl_3d_feature_getset_t [...]

Seems correct to me please verify below condition in your code
if (application != nullptr)
{
    // app_name is of type: const char* app_name
    get_3d_property.ApplicationName = const_cast<char*>(app_name);
    get_3d_property.ApplicationNameLength = static_cast<int8_t>(strlen(app_name));
}

is it restricting to populate application name in the structure?

please refer below Screen-shot 

[image]

your structure should be populated as mentioned in above screen-shot

paulober, 2023-01-23:
> Seems correct to me please verify below condition in your code if (application != nullptr) { // app_name is of type: const char* app_name get_3d_property.ApplicationName = [...]

The condition seems strange because it originally looks like this:
```cpp
if (application != nullptr)
  {
      std::string app_name = marshal_as<std::string>(application);
      // +1 for the NULL terminator
      char* pBuf = new char[app_name.length() + 1];
      strcpy_s(pBuf, app_name.length() + 1, app_name.c_str());
      set_3d_property.ApplicationName = pBuf;
      set_3d_property.ApplicationNameLength = sizeof pBuf;
  }
```
`application` is of type `String^` thats why this conversion is happening but as you can see in the debuger its working:

[image]

paulober, 2023-01-23:
But I still get `-547470120` error code...

paulober, 2023-01-24:
But when i set it I get SUCCESS for the set function call

paulober, 2023-01-24:
Ok, so if i replace my `ApplicationNameLength` code with:
```cpp
get_3d_property.ApplicationNameLength = static_cast<int8_t>(strlen(pBuf));
```
it works.

closed 2023-01-24 by paulober