intel/drivers.gpu.control-library #40: ctlEnumerateDisplayOutputs returns CTL_RESULT_ERROR_INVALID_SIZE

c3martin, 2023-02-08:
The first call to ctlEnumerateDisplayOutputs returns 3 display outputs, however sometimes the second call to ctlEnumerateDisplayOutputs with pCount set to 3 returns CTL_RESULT_ERROR_INVALID_SIZE.

See example below:

```
function main( ... )
# Get the adapter handles in hDevices as shown in the IGCL API programming guide
for(index = 0 .. Adapter_count-1)
     ctl_display_output_handle_t *hDisplayOutput = nullptr;
     # Enumerate all displays
        uint32_t Display_count = 0;
        result = ctlEnumerateDisplayOutputs(hDevices[Index], &Display_count, hDisplayOutput);
        // Display_count = 3
        // Result = CTL_RESULT_SUCCESS

        hDisplayOutput = (ctl_display_output_handle_t *)malloc(sizeof(ctl_display_output_handle_t) * Display_count);
        result = ctlEnumerateDisplayOutputs(hDevices[Index], &Display_count, hDisplayOutput);
        // Display_count = 3
        // Result = CTL_RESULT_ERROR_INVALID_SIZE
```

rahulrgandhi99, 2023-04-18:
> The first call to ctlEnumerateDisplayOutputs returns 3 display outputs, however sometimes the second call to ctlEnumerateDisplayOutputs with pCount set to 3 returns CTL_RESULT_ERROR_INVALID_SIZE.
> 
> See example below:
> 
> ```
> function main( ... )
> # Get the adapter handles in hDevices as shown in the IGCL API programming guide
> for(index = 0 .. Adapter_count-1)
>      ctl_display_output_handle_t *hDisplayOutput = nullptr;
>      # Enumerate all displays
>         uint32_t Display_count = 0;
>         result = ctlEnumerateDisplayOutputs(hDevices[Index], &Display_count, hDisplayOutput);
>         // Display_count = 3
>         // Result = CTL_RESULT_SUCCESS
> 
>         hDisplayOutput = (ctl_display_output_handle_t *)malloc(sizeof(ctl_display_output_handle_t) * Display_count);
>         result = ctlEnumerateDisplayOutputs(hDevices[Index], &Display_count, hDisplayOutput);
>         // Display_count = 3
>         // Result = CTL_RESULT_ERROR_INVALID_SIZE
> ```
> 
> Given the above, my questions are:
> 
> 1. In what circumstance would ctlEnumerateDisplayOutputs return CTL_RESULT_ERROR_INVALID_SIZE?
--> This error typically happens when you pass wrong input size.
> 2. Why would ctlEnumerateDisplayOutputs tell me that there are three display outputs and then fail on the second call with CTL_RESULT_ERROR_INVALID_SIZE?
--> What is your use case? Are you in b/w connecting or dis-connecting monitors.

ashutosh-d-shukla, 2023-04-27:
@c3martin any updates on the above issue?

c3martin, 2023-05-11:
@ashutosh-d-shukla @rahulrgandhi99 

Here are some more details on the issue:

Description:
The second call to ctlEnumerateDisplayOutputs fails when the number of displays changes during the application lifetime.

To demonstrate the issue, here a zip containing a slightly modified version of the [I2C_AUX_Sample_App](https://github.com/intel/drivers.gpu.control-library/tree/master/Samples/I2C_AUX_Samples):

[intel_test_app.zip](https://github.com/intel/drivers.gpu.control-library/files/11449950/intel_test_app.zip)

Reproduction Steps:
1.	Ensure there is only 1 monitor connected to the target machine
2.	Run the application as a Windows Service (using a tool like [https://nssm.cc/](https://urldefense.proofpoint.com/v2/url?u=https-3A__nssm.cc_&d=DwMFAg&c=7dfBJ8cXbWjhc0BhImu8wVIoUFmBzj1s88r8EGyM0UY&r=h6dwb4vggIYU3qdJPg-n1Iv4ZrESJunMcbRHyRSSjzk&m=ZI6o-ByohoUIn-Eg5d5SNdS-8CQHBsa1kYbfWYdsv4hYkDew925RvNe9Nmpil3ce&s=4nz2Ood_ACGn-EmHQlWj_33vXuywq0axKGpCv40dZsE&e=))
3.	Connect additional display(s) while the application is still running

**Note. This issue only occurs when the application is run as a Windows Service, running as a user application does not reproduce this issue.**

Logs:
// Single Display (Laptop Screen)
```
...
6) Scanning for Displays...
Adapter 0) Display Count: 1
```

// Connect 2 additional displays (3 displays in total)
```
7) Scanning for Displays...
Adapter 0) Display Count: 3
ctlEnumerateDisplayOutputs returned failure code: 0x4000000F
8) Scanning for Displays...
Adapter 0) Display Count: 3
ctlEnumerateDisplayOutputs returned failure code: 0x4000000F
```

ashutosh-d-shukla, 2023-06-08:
We are trying to replicate internally for this and will get back with the update.

ashutosh-d-shukla, 2023-09-17:
Tagging @murali-i as FYI, this could be fixed in one of the latest drivers as we found a memory corruption in driver that could be causing enumDisplayOutputs failure. Will post the driver version once we have it.

ashutosh-d-shukla, 2023-11-09:
This is expected to be fixed in V172 which is yet to be released externally in drivers at intel.com.

ashutosh-d-shukla, 2024-02-27:
Closing this as public drivers are released with this version. Please test with 31.0.101.5333 version from intel.com.

closed 2024-02-27 by ashutosh-d-shukla