intel/drivers.gpu.control-library #44: Multi GPU support

Tellusim, 2023-03-03:
ctlPowerTelemetryGet() returns telemetry of the last device.
It's impossible to get telemetry from the first device.
Reproduction app: https://gravitymark.tellusim.com/

Name: Intel(R) Arc(TM) A770 Graphics
Device: PCI\VEN_8086&DEV_56A0&SUBSYS_60011849&REV_08\6&2F5DA754&0&00080008
Memory: 8.00 GB
Screens: 1

Intel(R) Arc(TM) A770 Graphics // ctlGetDeviceProperties
Frequency: 600.00 MHz // ctlGetDeviceProperties
Temperature: 56 C
Utilization: 0 %
FanSpeed: 1135
Power: 18.22 W

Name: Intel(R) Arc(TM) A380 Graphics
Device: PCI\VEN_8086&DEV_56A5&SUBSYS_60041849&REV_05\6&B74F73D&0&00080009
Memory: 6.00 GB
Screens: 0

Intel(R) Arc(TM) A380 Graphics // ctlGetDeviceProperties
Frequency: 600.00 MHz // ctlGetDeviceProperties
Temperature: 56 C
Utilization: 0 %
FanSpeed: 1135
Power: 17.83 W

ezhilintc, 2023-04-14:
Please refer to this sample application which uses `ctlPowerTelemetryGet`. `ctlEnumerateDevices` can be used to enumerate the adapters and passing correct adapter handle to `ctlPowerTelemetryGet` will fetch the corresponding adapter's telemetry. 

https://github.com/intel/drivers.gpu.control-library/blob/master/Samples/Telemetry_Samples/Sample_TelemetryAPP.cpp

Please verify and confirm if this resolves this issue and this can be closed.

ashutosh-d-shukla, 2023-04-18:
@Tellusim please confirm if your query is resolved.

ashutosh-d-shukla, 2023-04-27:
@Tellusim any updates on this from your end?

Tellusim, 2023-04-27:
Sorry for the delay with the answer. The Telemetry_Samples.exe reports the same wrong results as GravityMark.
If you run the benchmark on a secondary GPU, the "Print Telemetry" from Telemetry_Samples.exe will show the first GPU counters for both GPUs.
GPU-Z shows correct sensor information. It was tested on the 31.0.101.4314 driver.

ashutosh-d-shukla, 2023-07-20:
@ezhilintc can we get this prioritized and chekced? There is same issue opened in our internal repo as well: https://github.com/intel-innersource/drivers.gpu.control-library/issues/1327

ashutosh-d-shukla, 2023-11-09:
@ezhilintc can we prioritize and fix this issue.

ezhilintc, 2024-07-03:
31.0.101.5594 driver should have this fix.

closed 2024-07-03 by ashutosh-d-shukla