intel/drivers.gpu.control-library #52: ctlInit returns CTL_RESULT_ERROR_UNSUPPORTED_VERSION

1Cherish1, 2023-05-04:
I got a CTL_RESULT_ERROR_UNSUPPORTED_VERSION return value when I call to ctlInit. IGCL version: 1.1，CPU: i7-12700，GPU: UHD Graphics 770(graphics drivers have been updated to latest version). I've seen the version of Controllib.dll in /System32. it's 1.0.96.0. I wonder how can I solve this problem? Is my hardware compatible with IGCL1.1? Thanks.

1Cherish1, 2023-06-16:
Thanks！Graghics driver version is 31.0.101.3169，I tried IGCL1.1's Generic Sample and CombineDisplay Sample，there's still such a problem. There won't be the same problem when I try 1.0' samples

ashutosh-d-shukla, 2023-11-09:
Can you please try the latest drivers 31.0.101.4952 from intel.com and confirm if issue is still seen.

Valkirie, 2024-01-28:
Similar to https://github.com/intel/drivers.gpu.control-library/issues/71 and https://github.com/intel/drivers.gpu.control-library/issues/70

ashutosh-d-shukla, 2024-02-27:
Issue #71 and #70 are different. This issue is specific to i7-12700 which is a supported platform. Have you tried the latest drivers from intel.com and confirmed if issue is still seen?

ashutosh-d-shukla, 2024-07-03:
Closing this as the reported problem is expected due to unsupported platform, please feel free to reopen if you can share more details for us to investigate.

closed 2024-07-03 by ashutosh-d-shukla