intel/drivers.gpu.control-library #61: return CTL_RESULT_ERROR_KMD_CALL  in edid management

Petercg, 2023-09-05:
I'm trying to get edid infomation ,but everytimes I run the app, it returns the log :CTL_RESULT_ERROR_KMD_CALL
Could you help me solve this problem?

Display :Intel(R) UHD Graphics 710（version:31.0.101.4688）
Processor : Intel(R) Pentium(R) Gold G7400 3.70GHz

ashutosh-d-shukla, 2023-12-01:
@Petercg this specific feature was enabled for your system some time in WW36/37 only so probably you tested on the older driver. Can you please test with the latest drivers from intel.com: 101.4887 or 101.4953 and confirm if issue is resolved.

ashutosh-d-shukla, 2023-12-01:
Tagging @saumyajyoti-mukherjee here as well.

ashutosh-d-shukla, 2024-02-27:
No updates, resolving this with latest drivers: 101.4887 or 101.4953. Please reopen if you see issue with the latest drivers from intel.com.

closed 2024-02-27 by ashutosh-d-shukla

yannickModulo, 2024-07-16:
Hi,

I have the same issue with a NUC13ANB-M (ASus) with windows 111 24H2 and latest driver 32.0.101.5762. is edid management working on  NUC?