intel/drivers.gpu.control-library #71: ctlInit returns CTL_RESULT_ERROR_UNSUPPORTED_VERSION

songpeng22, 2024-01-18:
I got a CTL_RESULT_ERROR_UNSUPPORTED_VERSION return value when I call to ctlInit. IGCL version: 1.1，CPU: intel x6413E，GPU: Intel® UHD Graphics for 10th Gen Intel® Processors. I've copied ControlLib32.dll from https://software.intel.com/sites/downloads/symbols/ControlLib32.dll/6552F6A939000/ControlLib32.dll , which I found it latest one（14-Dec-2023 19:27）, and put it in application path. I also tried one from C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_0f12908b4ff13ff7\, but some result. I wonder how can I solve this problem? Is my hardware compatible with IGCL1.1? Thanks.

Valkirie, 2024-01-24:
Would also love to get IGCL working on IceLake: https://github.com/intel/drivers.gpu.control-library/issues/70

Valkirie, 2024-02-08:
@murali-i any idea ?

ashutosh-d-shukla, 2024-02-27:
Control library is not supported on 10th Generation core processors and is supported from 12th Gen core processors onwards.

closed 2024-02-27 by ashutosh-d-shukla