intel/drivers.gpu.control-library #78: CTL_RESULT_ERROR_OS_CALL on ctlGetDeviceProperties

Valkirie, 2024-03-01:
When calling ctlGetDeviceProperties on an already running application, where IGCL has been initialized, and after a display adapters has been turned off then turned on, ctlGetDeviceProperties will return CTL_RESULT_ERROR_OS_CALL on said device.

kpsam2000Intel, 2024-04-04:
Hi @Valkirie... do you close the existing DX adapter/device as well during such an additional GPU plug-in event? If not can you try it and let us know? Thanks

Valkirie, 2024-04-08:
> Hi @Valkirie... do you close the existing DX adapter/device as well during such an additional GPU plug-in event? Close everything including IGCL using ctlClose and recreate adapter/IGCL. Can you try it and let us know? Thanks

Hey @kpsam2000Intel, sorry I'm slightly confused by what you mean with close the existing DX adapter/device. Whenever a new device is added, I'm closing and (re)opening IGCL. Please see GetDeviceIdx() in the following c# wrapper I made : https://pastebin.com/neu3rK05

Valkirie, 2024-09-26:
@kpsam2000Intel still around ?

ashutosh-d-shukla, 2024-09-29:
I believe Sameer was referring to disabling the current adapter that was enabled as well, will let him confirm but please try disabling that device before IGCL close and see if that helps.