intel/drivers.gpu.control-library #79: ctlInit returned failure code: 0x40000026 CTL_RESULT_ERROR_LOAD

mp437376671, 2024-03-04:


mp437376671, 2024-03-04:
Where can I get the shit ControlLib.dll? Can you explain clearly the situation of each error code? Why not directly provide the 64 bit ControlLib.dll library file and the 32-bit ControlLib32.dll, or provide the DLL source code

mp437376671, 2024-04-09:
Okay, I tried debugging on my own computer with a CPU i5-10200H. I tried downloading and installing driver versions 31.0.101.2121 and 31.0.101.2125 respectively, but they both failed and returned CTL_RESULT-ERROR-LOAD because ControlLib.dll is not available in my System directory. So I tried to port the project to another laptop in my company, with a CPU of I5-12500H and a driver version of 31.0.101.4575. It ran successfully and there were no errors. ControlLib.DLL is located in the System directory, so I can confirm that it is the reason why the CPU is not supported.

Meldryt, 2024-09-07:
@mp437376671 
Do u know where do i get ControlLIb.DLL?
Is it part of Intel Graphics Driver ?

ashutosh-d-shukla, 2024-09-29:
[Intel® Core™ i5-10200H Processor](https://ark.intel.com/content/www/us/en/ark/products/208016/intel-core-i5-10200h-processor-8m-cache-up-to-4-10-ghz.html) is not a formally supported processor (Cometlake) for this library and hence not shipped in the drivers. While a simple load may have worked by copying, there is no guarantee on API support.

closed 2024-09-29 by ashutosh-d-shukla