intel/drivers.gpu.control-library #88: Edid management error

yannickModulo, 2024-07-16:
Hi,

we plan to use the Gpu control library to read and write EDID. I am trying to use the EDID sample but all call to ctlEdidManagement return CTL_RESULT_ERROR_KMD_CALL
I am using a 13th gen NUC (asus) with latest driver (NUC13 ANB-M with gpu driver 32.0.101.5762) 
I guess it's the same issue  : https://github.com/intel/drivers.gpu.control-library/issues/61

saumyajyoti-mukherjee, 2024-07-18:
Hi @yannickModulo ,
Can you please elaborate on what is the purpose here when you mention write EDID? Do you want to override Monitor EDID? This API does not provide the option to write any physical EDID as such. For reading the regular i2c access API can be used though.

yannickModulo, 2024-07-18:
Hi,
we want to be able able to read incoming EDID from a monitor, Video projector and save as a bin file. 
And the most important feature is to be able to force the EDID on the output to simulate a screen with a specific resolution and frequency. We use this with AMD Radeon PRO and Nvidia Quadro to emulate an edid on a output (it should works with no screen connected on the output or with a screen connected).  See here for example  : https://www.nvidia.com/content/Control-Panel-Help/vLatest/en-us/mergedProjects/nvwks/Manage_EDID.htm
I suppose that your sample Edid-management_Sample is to do this but I have an error to read EDID, override or lock EDID. I have done test today with an ARC A770 and I have the same issue. 
Perhaps I don't understand how to use the sample. For example what is the syntax to force the EDID with a bin edid file on a specific output ?

saumyajyoti-mukherjee, 2024-07-18:
Hi @yannickModulo,
Thanks for the clarification. Yes, the use case you mentioned can be achieved using this API. 
This feature is not enabled by default as it comes with few side effects. e.g. Multi-plane overlay will get disabled. So, all Desktop composition will be done by OS.
You need to set below regkey _**EdidMgmtEnable= 1**_ , in Driver inf path under HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\control\class\{guid}\xxxx . 
This API is supported AlderLake platform onwards. NUC 13 Gen should be supported. To confirm can you share your Intel Gfx device HW ID from the device manager\Display Adapters tree?

yannickModulo, 2024-07-18:
Hi 

at least on the ARC A770 it works fine :)  is there a way to query from your API or any WIN32 API the "class{guid}\xxxx" to automatically install the registry key ?

saumyajyoti-mukherjee, 2024-07-19:
Hi @yannickModulo,
Thanks. You can try the below PowerShell routines from an admin prompt.
```powershell
function SetAdapterRegKey($Name, $value) {
    $devices = Get-PnpDevice -PresentOnly -Class 'Display' -FriendlyName 'intel*'

    foreach ($dev in $devices) {
        <# $dev is the current item #>
        
        $keypath = ((Get-PnpDeviceProperty -InstanceId $dev.DeviceID) | Where-Object KeyName -Like 'DEVPKEY_Device_Driver').Data
        Write-Output  $keypath " :: " $Name " = " $value
        $regPath = 'HKLM:\SYSTEM\CurrentControlSet\Control\Class\' + $keypath
        New-ItemProperty -Path $regPath -Name $Name -PropertyType DWord -Value $value -Force | Out-Null
    }
}

function EdidMgmtSupport {
    <#
     .Synopsis
     Set adapter reg keys for EDID management Support on/off.
     #>
    param (
        [Parameter(Mandatory)]
        [string]
        [ValidateSet('Enable', 'Disable')]
        <# State of Edid Management Feature Support. #>
        $State
    )
    if ($State -eq 'Enable') {
        SetAdapterRegKey "EdidMgmtEnable" 1
    }
    else {
        SetAdapterRegKey "EdidMgmtEnable" 0
    }
}
```

saumyajyoti-mukherjee, 2024-07-19:
As the issue mentioned in this issue is addressed, can we close the ticket?

closed 2024-07-19 by yannickModulo