intel/drivers.gpu.control-library #91: Enumerate physical available outputs

yannickModulo, 2024-07-19:
Hi,

I need to be able to list the available outputs of the graphics card. When I use CltEnumerateDisplayOutput, i have 25 outputs listed. It's easy to find the connected output with CTL.....DISPLAY_ACTIVE and CTL....DISPLAY_ATTACHED. but I don't find a way to know which of the 25 outputs is physically available

saumyajyoti-mukherjee, 2024-09-02:
Hi @yannickModulo, currently there is no direct correspondence between OS Display list vs IGCL list. The Target ID will be same. The physical port also cannot be guaranteed to be in same numbered order across different designs. So, even though the IGCL lists item in order of Physical port numbers, it may not be in same physical order in board. 
Some display properties like output type/Connector types can be used to get some idea on the display type.

saumyajyoti-mukherjee, 2024-09-03:
The OS aware Target ID (QueryDisplayConfig returned value) will match the Intel Control API display properties returned handle Os_display_encoder_handle, as mentioned in previous comment also. 
Additionally, you may use OS API to query desktop relative coordinates to get an idea on the target id and the corresponding monitor.

ashutosh-d-shukla, 2024-09-29:
@yannickModulo let us know if the suggestions have helped to resolve your query.