intel/drivers.gpu.control-library #92: EDID and Lane count and bandwidth

yannickModulo, 2024-07-24:
Forcing EDID is working fine up to 1920x1200 resolution but for more important resolution, the EDID is correctly set but the real resolution is limited to 1920x1200. on other gpu I have function to set the number of lane and bandwidth of each lane on the displayport. I suppose I need to force more lane and the bandwidth but it seems there is no API to do this ?

saumyajyoti-mukherjee, 2024-09-02:
Hi @yannickModulo, Which target are you trying to override? If a physical Target is present, it will limit the modes by DPCD Caps etc. If no read target is present and you try to override it should work. Can you please clarify?

yannickModulo, 2024-09-02:
I want  to force the EDID on a physical target and force an EDID. For example even if I plus an HD monitor, I want to be able to force a 4K EDID. But because there is no function to force the bandwith and the number of lane, when I do this,  it don't works. On other GPU brand there is a function to force the bandwidth and number of lane to allow the correct bandwidth for a bigger resolution than HD.

saumyajyoti-mukherjee, 2024-09-17:
Currently usecase planned for EDID management API is to lock EDID when changing monitor to avoid config switches or modify some resolutions in EDID. So, when panel is not there it can work with any resolution as it does not assume any DPCD values. But when real monitor is plugged in we assume Monitor DPCD for BW limits.

ashutosh-d-shukla, 2026-02-28:
Closing this as there is no response for more than 5 months.

closed 2026-02-28 by ashutosh-d-shukla