intel/drivers.gpu.control-library #95: C++ program to fetch the Intel Arc GPU and VRAM usage

vamshigowda56vg-git, 2024-09-04:
can anyone provide an example C++ program to fetch Intel Arc GPU and VRAM usage?

ashutosh-d-shukla, 2024-09-29:
Please refer to https://github.com/intel/drivers.gpu.control-library/blob/master/Samples/Overclocking_Sample/Sample_OverclockAPP.cpp for reference. Resolving this. If specific queries are there feel free to reopen. Tagging @ezhilintc as FYI

closed 2024-09-29 by ashutosh-d-shukla

ezhilintc, 2024-09-29:
Refer below functions and the API usages in this sample application
https://github.com/intel/drivers.gpu.control-library/blob/master/Samples/Telemetry_Samples/Sample_TelemetryAPP.cpp
| function | API to get usage | 
|-|-|
| CtlMemoryTest | ctlMemoryGetState |
| CtlEngineTest | ctlEngineGetActivity |