intel/drivers.gpu.control-library #117: Telemetry is actually available for 32-bit.

17314642, 2025-04-27:
Is it error in readme or there was an update?

```
Telemetry Success
 

TimeStamp: 11793.196293 (Time in Seconds) Datatype:(DOUBLE)

Gpu Energy Counter: 514648.529644 (Energy in Joules) Datatype:(DOUBLE)

Gpu Voltage: 0.650000 (Voltage in Volts) Datatype:(DOUBLE)

Gpu Current Frequency: 750.000000 (Frequency in MHz) Datatype:(DOUBLE)

Gpu Current Temperature: 54.000000 (Temperature in Celsius) Datatype:(DOUBLE)

Gpu Activity Counter: 219.391744 (Time in Seconds) Datatype:(DOUBLE)

Render Activity Counter: 53.382857 (Time in Seconds) Datatype:(DOUBLE)

Media Activity Counter: 0.000000 (Time in Seconds) Datatype:(DOUBLE)

VRAM Frequency: 2000.000000 (Frequency in MHz) Datatype:(DOUBLE)

VRAM Read Bandwidth Counter: 2482652563648 (Memory in Bytes) Datatype:(UINT64)

VRAM Write Bandwidth Counter: 600687131840 (Memory in Bytes) Datatype:(UINT64)

VRAM Effective Frequency: 16000.000000 (Frequency in MHz) Datatype:(DOUBLE)

VRAM Temperature: 56.000000 (Temperature in Celsius) Datatype:(DOUBLE)

Fan[0] Speed: 295.000000 (Angular Speed in RPM) Datatype:(DOUBLE)

gpuPowerLimited: false

gpuTemperatureLimited: false

gpuCurrentLimited: false

gpuVoltageLimited: false

gpuUtilizationLimited: false
```

17314642, 2025-09-18:
@pacampbe 32-bit sample app using 32-bit igcl library

pacampbe, 2025-09-19:
@17314642 In general, as the comment in the readme says, many of the functions in IGCL are simply loading the level0 DLL and forwarding the requests. For APIs which depend on level0, their asset is only built as a 64 bit DLL, so it can't be loaded by a 32 bit application. Looking at the implementation of the APIs in IGCL, used by the sample application -- it appears to me that instead of using level0, they are currently implemented where they go directly to the source for the information. Due to this, the API calls are able to work even on 32 bit applications. This could change in the future as it is not advertised to be supported, so I would not recommended to build a 32 bit application and expect this to work indefinitely.

17314642, 2025-09-21:
Okay, thanks for clarification.

Is there a way to gather metrics from 32-bit apps when using new intel gpus, such as Intel Arc A770 and later? My use-case is implementing an in-game hardware monitoring overlay. Or would I need to make a 64-bit shim which collects data an passes that info to 32-bit app?